geezacoleman/OpenWeedLocator #1: Incorporate GPS for RT speed compensation

geezacoleman, 2021-08-23:
Suggestion from Todd Matthews via Twitter to include FP9 GPS for turn compensation.

Fairly low cost parts: https://www.sparkfun.com/products/16475

Would require the addition of GPS comms/data retrieval class/functions and inclusion into delay/duration variables.

EchelonMirror, 2022-09-16:
Instead of a $300 GPS module you can use a $30 IMU (https://www.adafruit.com/?q=DOF&sort=BestMatch). These will give you fast and accurate speed measurements, as well as jerks in the speed. This can also allow for a form of turn compensation since it can measure the device moving in 3D space (yaw/pitch/roll) relative to other units on the boom.

geezacoleman, 2022-09-16:
That makes a lot of sense for more sensitive speed measurements, I had considered using IMUs for this.  Though since this issue was posted the benefit of weed locations/GPS data in image EXIF has become more clear so have shifted towards GPS alone. Thoughts on GPS alone for effective speed or would you need IMUs as well for accurate compensation?

EchelonMirror, 2022-09-16:
GPS will do the job for speed. Turn compensation more difficult since the devices have no real understanding of yaw movement if you have a single point measurement.