geezacoleman/OpenWeedLocator #8: STL files as STEP for modification?

rfirth-code, 2021-09-01:
Any chance we could get the .step files for the case and cover that are currently STL downloads?
My 3D printer can't handle the build volume so I was looking to modify for laser cutting instead. 
Awesome project!

rfirth-code, 2021-09-01:
@plains203 thanks for the link - do you know if their is any way of downloading this in a 3d format other than STL or OBJ?

plains203, 2021-09-01:
That I don’t know.

geezacoleman, 2021-09-01:
It looks like TinkerCAD doesn't support that conversion, but there are a couple of other ways mentioned [here](https://all3dp.com/2/stl-to-step-how-to-convert-stl-files-to-step/#:~:text=A%20straightforward%20method%20of%20converting,to%20upload%20the%20files%20to.&text=From%20there%2C%20you%20can%20select,to%20export%20to%2C%20including%20STEP.), using either Fusion360 or FreeCAD. Hopefully that helps!

If you do end up converting them, open up a pull request and we can add them to the 3D models folder too.

MerbeinRobotics, 2021-09-03:
Hi,  I have the same issue with the size of the cover and case, it's just a bit too big for my Creality Ender 3 3D printer.  But think I can modify the STL file through Fusion 360.  I'll have a go....

But I'm really excited about this project.  I've been working on an autonomous vineyard robot based on the Ardurover/Cube/Mission planner project and the OWL will be perfect for it.    I also wanted to get more familiar with the Raspberry Pi and computer vision and this is the perfect opportunity to do so.  

Well done!

Kind Regards,

Stephen

geezacoleman, 2021-09-06:
Thanks Stephen for the feedback! It will be great to see how you use this. Make sure you share the outcome once you're done too and any suggestions for improvements along the way.

Any luck modifying the STL on Fusion 360? @williamtsalter perhaps if you linked the original Tinkercad for the base then it could be ungrouped and more easily adjusted. This length was selected because it only just fit on my Prusa Mk3S.

williamtsalter, 2021-09-06:
@MerbeinRobotics and @geezacoleman, I just had a look at the public model on TinkerCad and you should already be able to ungroup the base for easy modification. It should become a colourful mess when you ungroup. If this isn't the case for you, please let me know as I may have to change the settings for other users.
[image]

geezacoleman, 2021-09-06:
Thanks @williamtsalter - I just tried that using the link above and the model comes apart nicely as in the image. Ctrl + shift + g to ungroup or there's a button in the top right of the frame.

MerbeinRobotics, 2021-09-07:
Thanks @geezacoleman and @williamtsalter.  After a bit of fiddling with Cura I discovered the case will fit in the Ender 3 if I print it vertically (with the end on the base).  Probably not ideal and I would prefer to print it from the base up.    I have  been able to convert the STL (mesh) file to a solid body in Fusion 360 ready for modification.  My initial thought was to rotate the camera mounts 90 degrees so that I can hopefully shorten the overall length by about 23mm.

geezacoleman, 2021-09-09:
That's good to hear @MerbeinRobotics that it fits - it is quite tall, but hopefully printing it vertically works. 

I'll close this issue now as we've addressed the original question, but open up a new issue if you end up shortening it and we can incorporate at short model into the set too.

closed 2021-09-09 by geezacoleman