geezacoleman/OpenWeedLocator #9: generate ROS nodes for OWL

geezacoleman, 2021-09-01:
Improve cross compatibility with the fairly widely accepted standard for ag robotics. I'm very unfamiliar with ROS (though really should be better!), so this is a good project if someone wants to take it on.

geezacoleman, 2022-03-22:
Using ROS nodes for the OWL would be a big improvement in functionality, modularity and repeatability, but could throw [some issues with Raspbian vs. Ubuntu](https://roboticsbackend.com/using-ros-on-raspberry-pi-best-practices/).

Will continue looking into it and changing the system over.

EchelonMirror, 2022-09-16:
We can have a chat about this in a couple of months and see how we can help with a ROS2 port (ROS2 is a heck of a lot better).

closed 2024-11-14 by geezacoleman

geezacoleman, 2024-12-16:
That would be great - let me know how it goes and if you have any questions. The main objective would be to allow OWL to be better integrated with field robots instead of just running as a simple python script.

RtJinxuWang, 2025-04-27:
> That would be great - let me know how it goes and if you have any questions. The main objective would be to allow OWL to be better integrated with field robots instead of just running as a simple python script.

Sorry, I was busy with my own precise spraying project some time ago, so I rewrote a system based on ros by myself. I currently only have an nvidia device. When I have time later, I will try to encapsulate OWL into ros on an nvidia device. Besides, I have relatively little knowledge about hardware. I would like to ask you, for example, what types of cameras do the precision spraying company bilberry in France and John Deere's see and spray use? I'm currently using an industrial camera, but it's too severely affected by light!

EchelonMirror, 2025-04-27:
What light issues are you having you having with the industrial camera?

RtJinxuWang, 2025-04-28:
> What light issues are you having you having with the industrial camera?

[image]
This picture has a particularly poor imaging effect in the shadow state

geezacoleman, 2025-04-30:
> Sorry, I was busy with my own precise spraying project some time ago, so I rewrote a system based on ros by myself. I currently only have an nvidia device. When I have time later, I will try to [...]

That's awesome - would be nice to see OWL on an NVIDIA device with ROS. Let me know how you go.

I'm unsure of the specifics of the hardware those companies use, and they unfortunately don't like to share and I also haven't had the opportunity to pull one apart! For research we have used FLIR Blackfly and LucidVision TritonS machine vision cameras. All struggle with harsh shadows unfortunately, which is why many companies invest in strobe or LED lighting.

The image above though, the information in the shadows is still there, so training a model on the image would likely still work quite well, it would just need data from images with shadows.

[image]