geezacoleman/OpenWeedLocator #14: Error with file logger

dtms6, 2022-01-07:
Having assembled an OWL and installed the software.I continue to receive the following error once having focused the camera:

FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220108-065504_weed_log.txt'

The error occurs almost instantaneously and kills the program.
There appears to be an issue with inability to write and then recall a file log in the logger directory.

I haven't altered any code and have reinstalled/updated the OWL software just in case, but the error remains.

Would appreciate some guidance. 

The output is as follows:

(owl) pi@owl:~ $ ~/owl/./greenonbrown.py
linux
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.
Solenoid 3 ON
Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 291, in hoot
    self.controller.receive(nozzle=i, delay=delay, timeStamp=sprayTime, duration=sprayDur)
  File "/home/pi/owl/relay_control.py", line 132, in receive
    self.logger.log_line(line)
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220108-065030_weed_log.txt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 417, in <module>
    minArea=10)
  File "/home/pi/owl/./greenonbrown.py", line 325, in hoot
    self.logger.log_line("[CRITICAL ERROR] STOPPED: {}".format(e))
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220108-065030_weed_log.txt'

geezacoleman, 2022-01-07:
Thanks for putting one together and for the heads up!

It looks like this might be an issue with the absolute vs. relative path references where it's looking for `logs/20220108-065030_weed_log.txt` from `(owl) pi@owl:~ $ ~/owl/./greenonbrown.py` where it's looking for that log file in the root directory.

If you try running it from the owl directory, does it work? 
`> cd owl` 
`> ./greenonbrown.py`

I'll make a quick change to the code to fix it if so.

dtms6, 2022-01-08:
Yes. That resolves the issue! Works a treat.
I'll share some photos of the unit.

geezacoleman, 2022-01-08:
Glad that worked out! I'll leave this issue open for the moment so I can fix it up next week.

Looking forward to see how the unit ended up.

geezacoleman, 2022-01-10:
@dtms6 this should now be fixed - you'll just need to update the owl folder with the latest version of the repo.

closed 2022-01-10 by geezacoleman

dtms6, 2022-01-10:
Yes it fixed the problem. Updated the files and works like a treat. Thanks.

PR #15 (2022-01-10): Fixes log file bug
Fixes #14