geezacoleman/OpenWeedLocator #27: adapt code for section control rather than nozzle control

geezacoleman, 2022-03-08:
instead of activating individual nozzles and requiring many solenoids, adapt the code so the weeds are detected over a single channel. This could be linked to solenoid on a section for coarser but lower cost use.

(suggested by a grower in WA wheatbelt)

EchelonMirror, 2022-09-16:
Just need to make sure the valve for the section control is fast enough to react when you're travelling at 3m/s to 6m/s. A lot of section control valves are pretty slow, new ones are relatively fast but the vast majority of situations they will be slow.

Completely doable. Just note large valve will need a slightly larger solenoid and larger current, but a 2A MOSFET will do the job.

Minorplanets76, 2023-05-14:
Section control is pretty slow as there is a pressure build up/bleed off associated especially when mounted back at the manifold.  It's fine for headlands and going around trees and stuff, but unless weeds were pretty sparse.  As I already have section control via AgOpenGPS, I'm heading in the opposite direction, to control each nozzle independently.  That way the existing section control takes care of coverage and OWL decides if there's a weed to spray.  A solenoid at the nozzle gives the response time, but they are pretty expensive.   AgOpenGPS has just started supporting 64 sections, but still doesn't like it would be enough for those WA farmers!