geezacoleman/OpenWeedLocator #51: Software issues focusing camera

manninb, 2022-09-07:
Hi,
I'm not able to get an image from the camera up after following the instructions, wondering if anyone can offer help. see below.

pi@raspberrypi:~ $ ps-C greenonbrown.py
bash: ps-C: command not found
pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  533 ?        01:03:14 greenonbrown.py
pi@raspberrypi:~ $ sudo kill 533
pi@raspberrypi:~ $ ~/owl/./greenonbrown.py --show-display
linux
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:285: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information.
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.

geezacoleman, 2022-09-07:
Thanks @manninb, based on this [thread](https://forums.raspberrypi.com/viewtopic.php?t=275387), it looks like it could be a Python2 vs Python3 issue for RPi.GPIO

```
sudo apt update
sudo apt install python3-rpi.gpio
```

Let us know how that goes - we will need to update the instructions/requirements.txt file if so!

manninb, 2022-09-07:
It didn't seem to work I put the script in the terminal window an it seemed to run. see below

I closed the unit down and restarted it but could still not get the camera vision.

do you have any other ideas or suggestions as to where I went wrong

pi@raspberrypi:~ $ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]      
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]        
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [392 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.6 MB in 47s (287 kB/s)                                              
Reading package lists... Done
Building dependency tree      
Reading state information... Done
196 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
pi@raspberrypi:~ $ sudo apt install python3-rpi.gpio
Reading package lists... Done
Building dependency tree      
Reading state information... Done
python3-rpi.gpio is already the newest version (0.7.0-0.1~bpo10+4).
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 196 not upgraded.
pi@raspberrypi:~ $

manninb, 2022-09-07:
Further to the above this is the script I get when I attempt get the camera view

pi@raspberrypi:~ $ ps -c greenonbrown.py
error: unknown user-defined format specifier "nbrown.py"

Usage:
 ps [options]

 Try 'ps --help <simple|list|output|threads|misc|all>'
  or 'ps --help <s|l|o|t|m|a>'
 for additional help text.

For more details see ps(1).
pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  526 ?        01:09:19 greenonbrown.py
pi@raspberrypi:~ $ sudo kill 526
pi@raspberrypi:~ $  ~/owl/./greenonbrown.py --show-display
linux
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:285: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. 
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.
Solenoid 0 ON
Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 285, in hoot
    self.controller.receive(nozzle=i, timeStamp=sprayTime, duration=sprayDur)
  File "/home/pi/owl/relay_control.py", line 132, in receive
    self.logger.log_line(line)
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220907-160300_weed_log.txt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 406, in <module>
    minArea=10)
  File "/home/pi/owl/./greenonbrown.py", line 319, in hoot
    self.logger.log_line("[CRITICAL ERROR] STOPPED: {}".format(e))
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220907-160300_weed_log.txt'
pi@raspberrypi:~ $ ^C

geezacoleman, 2022-09-07:
Thanks for trying that - I think I've worked out why the `PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'` error is occurring. While it is just a warning from gpiozero, the greenonbrown.py script will only start a PiCamera instance if it can find RPi.GPIO, otherwise it defaults to a USB webcam, which is why the camera isn't starting.

This error PinFactoryFallback warning occurs when RPi.GPIO isn't installed - in this case it looks like it isn't installed in the `owl` virtual environment. Assuming you've already made an `owl` environment (looks like that's the case), you'll need to type the following commands:
```
workon owl
pip3 install rpi.gpio

```
This is based on a comment in the [gpiozero docs](https://gpiozero.readthedocs.io/en/stable/faq.html#why-do-i-get-pinfactoryfallback-warnings-when-i-import-gpiozero) FAQs about this warning.

You can also do:
```
workon owl
pip install -r requirements.txt
```
This error should have been fixed previously - make sure you're using the most up to date version of owl, by re-cloning the repository. If that doesn't work, to overcome the `FileNotFoundError` you'll need to launch `greenonbrown.py` from the `owl` directory. 

So if you:
```
cd owl
~/owl/./greenonbrown.py --show-display
```

geezacoleman, 2022-09-08:
Has this fixed the camera/logging issues? If so I'll close this issue.

Wifi is probably better addressed by the wider Raspberry Pi community, there could be a few different reasons for that.

manninb, 2022-09-12:
thank, I have got wi fi working for the time being ran the the last lot of script.

pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ pip3 install rpi.gpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi.gpio
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp37-cp37m-linux_armv7l.whl (47 kB)
     
Installing collected packages: rpi.gpio
Successfully installed rpi.gpio-0.7.1
WARNING: You are using pip version 21.1.2; however, version 22.2.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ pip install -r requirements.txt
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
WARNING: You are using pip version 21.1.2; however, version 22.2.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~ $
(owl) pi@raspberrypi:~ $ cd owl
(owl) pi@raspberrypi:~/owl $ ~/owl/.greenonbrown.py --show-display
bash: /home/pi/owl/.greenonbrown.py: No such file or directory
(owl) pi@raspberrypi:~/owl $

Attempted the camera view procedure again (twice) but was unable to get a view of the camera.

got these results

pi@raspberrypi:~ $ ~/owl/./greenonbrown.py --show-display
linux
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.

second time

pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  522 ?        00:05:39 greenonbrown.py
pi@raspberrypi:~ $ sudo kill 522
pi@raspberrypi:~ $ ~/owl/./greenonbrown.py --show-display
linux
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.
Solenoid 3 ON
Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 285, in hoot
    self.controller.receive(nozzle=i, timeStamp=sprayTime, duration=sprayDur)
  File "/home/pi/owl/relay_control.py", line 132, in receive
    self.logger.log_line(line)
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220912-104134_weed_log.txt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 406, in <module>
    minArea=10)
  File "/home/pi/owl/./greenonbrown.py", line 319, in hoot
    self.logger.log_line("[CRITICAL ERROR] STOPPED: {}".format(e))
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20220912-104134_weed_log.txt'
pi@raspberrypi:~ $ 

Would it be possible to have a chat about this sometime, It might save you some time.

geezacoleman, 2022-09-16:
As discussed on Wednesday, make sure you follow the [Update OWL procedure](https://github.com/geezacoleman/OpenWeedLocator#updating-owl) so you have the newest version of the software without this bug. I've raised an issue to release a new software version for download so this is much easier.

When pip installing the requirements.txt file you'll also need to change directory into the owl folder, so:

```
workon owl
cd owl
pip install -r requirements.txt
```

manninb, 2022-09-16:
Thanks, I seemed to be able to  update software, but the owl didn't trigger and I couldn't get the camera up. I re imaged a card from the github site and the unit worked but couldn't get the camera (see script below).
Is the version you get from the site different to the one you get by updating?

pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  528 ?        00:12:56 greenonbrown.py
(owl) pi@raspberrypi:~ $ sudo kill 528
(owl) pi@raspberrypi:~ $ ~/owl/./greenonbrown.py --show-display
linux
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:290: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  'Falling back from %s: %s' % (name, str(e))))
/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/gpiozero/devices.py:285: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information.
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
[INFO] Setting up nozzles...
Solenoid 0 OFF
Solenoid 1 OFF
Solenoid 2 OFF
Solenoid 3 OFF
[INFO] Setup complete. Starting spraying.
Solenoid 0 ON
Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 285, in hoot
    self.controller.receive(nozzle=i, timeStamp=sprayTime, duration=sprayDur)
  File "/home/pi/owl/relay_control.py", line 132, in receive
    self.logger.log_line(line)
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20210708-103148_weed_log.txt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/owl/./greenonbrown.py", line 406, in <module>
    minArea=10)
  File "/home/pi/owl/./greenonbrown.py", line 319, in hoot
    self.logger.log_line("[CRITICAL ERROR] STOPPED: {}".format(e))
  File "/home/pi/owl/logger.py", line 17, in log_line
    with open(self.saveDir, 'a+') as file:
FileNotFoundError: [Errno 2] No such file or directory: 'logs/20210708-103148_weed_log.txt'
(owl) pi@raspberrypi:~ $

manninb, 2022-09-16:
ps can you tell me the function of the commands below?

cd owl
pip install -r requirements.txt

geezacoleman, 2022-09-16:
The version downloaded from the site is the one that needs updating as mentioned above. It was the original software we had when the OWL was first launched but it is old and has lots of flaws. It works if you run greenonbrown.py from the OWL directory but doesn't if you run it from anywhere else. So I wouldn't redownload/reimage any cards again, just use this one. The version from the download is about a year behind the one you get from updating it.

To get this working you need to :
1. get the latest software from GitHub and save it to a new owl folder on the card, this is what I mean by 'updating'. This is done by following the [Update OWL procedure](https://github.com/geezacoleman/OpenWeedLocator#updating-owl).
2. This procedure guides you through installing all the required libraries/supporting software to avoid the rpi.gpio issue above.
3. Once you have carefully followed this you should have everything working fine.

`pip install -r requirements.txt` installs all the listed packages within the requirements.txt file. It's a faster way of adding all the various packages needed.

`cd` means change directory. It moves into the folder you specify. E.g. `cd owl` moves you into the owl folder if it exists.

geezacoleman, 2022-10-05:
Were you able to fix this @manninb? I might close this if so.

manninb, 2022-10-05:
Thanks, I have been called away to other jobs, but will be back onto this, so far no luck.

manninb, 2022-10-07:
thanks, I have got this to work, This is probably a dumb question but  can you give me the command to go from 
(owl) pi@raspberrypi:~/owl $
back to 
(owl) pi@raspberrypi:~ $

geezacoleman, 2022-10-07:
Ok great! What did you have to do differently?

Sure thing, just type:
`(owl) pi@raspberrypi: $ cd ..` (should have a space between the `cd` and the two `.`)

manninb, 2022-10-07:
I'm not sure exactly, I may not have been going back to the raspberry from the owl directory if that makes sense. I will try a few other units and see how it goes. There was quite a bit of interest in the OWL at the Nowley field day.

geezacoleman, 2022-10-07:
That's good to hear thanks for talking about it - feel free to write up a summary in the discussions page for any feedback/lessons learned etc.

I'll close this issue now as it's been resolved.

closed 2022-10-07 by geezacoleman