geezacoleman/OpenWeedLocator #53: Starting OWL on boot

MerbeinRobotics, 2022-09-13:
Hello, I am trying to get OWL to run on startup but I'm not having any luck so far so I would like to check my procedure, based on the software installation instructions is correct.  (I'm not very familiar with Linux so this is a great learning experience)

First, in the terminal window I create the virtual owl environment.
      $ mkvirtualenv owl -p python3

Then I make the greenonbrown.py file executable.

   (owl) (owl) pi@raspberrypi:~ $ chmod a+x ~/owl/greenonbrown.py

Next I bring up the crontab file to add the following line (although I note it is already listed).

    @reboot /home/pi/owl/owl_boot.sh
  
I save and exit the editor

Then I type the following line into the terminal

   (owl) pi@raspberrypi:~ $ chmod a+x ~/owl/owl_boot.sh

However it still goes to the Pi OS where I have to open the greenonbrown.py file to start detecting,  (which is working perfectly)

Is there something I am missing in the procedure?

Thanks for the help.

Kind Regards,

Stephen

geezacoleman, 2022-09-16:
Interesting! Apologies for the delay - was out at a few field days. I've had this issue before and it's usually been the result of the owl_boot.sh file or greenonbrown.py file not being executable or reachable.

If you want to double check if the files are executable, right click and open file properties ([as described here](https://askubuntu.com/questions/484718/how-to-make-a-file-executable)), alternatively type `ls - l` in the owl directory and it will list all the permissions on the left hand side for every file in the directory.

[image]

You may also want to try `sudo chmod a+x` on both greenonbrown.py and owl_boot.sh. It may be a user permissions thing - if so I'll need to update the instructions.

MerbeinRobotics, 2022-09-16:
Thanks @geezacoleman.  I was at the Riverland Field day myself yesterday.

I checked the properties for both greenonbrown.py and owl_boot.sh and as shown in the photo below they both say "Anyone" in the execute permissions box.

[image]

Anyway I decided to download the latest owl image which seemed to work ok.  However it still won't start on boot and goes to the desktop.  But when I "execute in terminal" greenonbrown.py it opens, one window, gives a beep and then closes.

Should I try downloading the image again?

Thanks for your time and help.

Much appreciated.

Stephen

geezacoleman, 2022-09-17:
Interesting - sorry this hasn't been an easy process for you!

There are some issues with the only OWL.img that's available here (as being discussed over on #51. So to overcome those I'd recommend using that image and just updating it with the [Updating Procedure](https://github.com/geezacoleman/OpenWeedLocator#updating-owl). I need to upload a new image (as mentioned in #54) to avoid these errors.

Let me know if that sorts it out for you!

MerbeinRobotics, 2022-09-17:
Hi @geezacoleman,

I went through the updating procedure but still unable to start greenonbrown.py.

When I reboot the RPi it gives a double beep from the buzzer and goes to the desktop.   When I "execute in terminal" greenonbrown.py the buzzer beeps once, a window opens but then closes again.

Also, when I type ls - l, I see

[image]

Just thought I'd share a few more clues.

Kind Regards,

Stephen

MerbeinRobotics, 2022-09-18:
Hi @geezacoleman.  Some progress,  I went back and re downloaded the OWL image and re flashed the image to the SD card and the weed detection started automatically without having to launch "execute in terminal" greenonbrown.py from the desktop.  Unfortunately I seem to have lost the abilty to look at the video screen, I just get the terminal screen listing which solenoids are on or off.

However when I did an update  it wouldn't start any detecting, either automatically on startup or by executing greenonbrown,py.

I'll reflash the image to the sd card and at least I'll be able to bolt it onto the tractor to do some real world testing.

Anyway, just updating my progress.

KInd Regards,

Stephen

geezacoleman, 2022-09-18:
> Hi @geezacoleman,
>
> I went through the updating procedure but still unable to start greenonbrown.py.
>
> When I reboot the RPi it gives a double beep from the buzzer and goes to the desktop. When I [...]

I think one of the issues here is that the software runs on boot and is still running in the background (given you're hearing two beeps). If you try running `greenonbrown.py` again from terminal without first stopping what's happening in the background, it won't be able to access the camera, because it's being used.

So if you want to run it again you need to stop the program running:
```
(owl) pi@raspberrypi:~ $ ps -C greenonbrown.py
PID TTY              TIME CMD
515 ?            00:00:00 greenonbrown.py

(owl) pi@raspberrypi:~ $ sudo kill enter_your_PID_number_here
```

This will allow you to stop the background operation of greenonbrown.py. Once you've done that, you can run it again with:
```
(owl) pi@raspberrypi:~ $ ~/owl/.greenonbrown.py --show-display
```

If you're using the old software, you'll need to open the greenonbrown.py python file and change `headless=False` to `headless=true`

MerbeinRobotics, 2022-09-18:
Hi again @geezacoleman 

Thanks for the previous reply.  I reflashed the owl image file and the detection starts on power up every time however I still can't get the video feed to work.  The photo below shows the procedure I used to try and get the video feed (after I stopped the program running). 

[image]

Another shot of the terminal output whenever I try to use the command line flags.
[image]

I suspect the problem is me and my lack of experience/understanding with RPi's. (one of the reasons why I like this project is to get more experience with RPi's)

Thanks for your time and patients.

Kind Regards,

Stephen

MerbeinRobotics, 2022-09-21:
Thanks @geezacoleman I will try the update and see what happens.  I'll take you up on the zoom meeting if I can't get the video feed sorted.  

In the mean time I can to some testing, including walking it around our vineyard.  Looking very promising so far.

[image]

Kind Regards,

Stephen

geezacoleman, 2022-09-21:
Glad to hear things are still looking promising despite these setbacks!

Email is guy.coleman[at]sydney.edu.au if you want to get in touch.

geezacoleman, 2022-10-05:
How did you go fixing this one @MerbeinRobotics - all sorted or still running into issues?

MerbeinRobotics, 2022-10-05:
Hi @geezacoleman I must admit I hadn't had a chance to to an update to see if the problem was solved but I tried tonight and I'm still having problems.  I'll send you an email to see if I can arrange a quick zoom meeting to get it sorted.  I strongly suspect the issue is with me so it'll probably take 5 minutes to sort out.

geezacoleman, 2022-10-13:
Thanks for the chat earlier @MerbeinRobotics. Just to summarise what we did:
* downloaded the new version of the software onto the old image
* ran into issues with RPi.GPIO not being installed within the `owl` environment. This was fixed by running `pip install RPi.GPIO`.
* image was very dark inside - the exposure compensation is turned down a lot for sunny conditions. Can make it look like the camera isn't working. Checking with a torch/lamp can help this.

This seemed to sort it out! Looking forward to seeing how it's used on your robot.

closed 2022-10-13 by geezacoleman

MerbeinRobotics, 2022-10-26:
Thankyou @geezacoleman Very much appreciated.

Kind Regards,

Stephen

Giridhar-Peepulagri, 2025-03-03:
i think you are directly implementing your image processing on the rpi itself.what 
if i use the google coral boards for the seperate image processing how can i connect it to the rpi 4