geezacoleman/OpenWeedLocator #57: Focusing

manninb, 2022-10-07:
Can you make any general comments on focusing. Should you just set heights so the field of view is 1m wide at the ground and then focus for the clear image of what's on the ground? Does focusing make much difference to the performance of the system?

geezacoleman, 2022-10-08:
The way we usually focus the camera is to firstly mount it at the height it will be used in the field and then adjust the focus ring (one is the aperture, and one is the focus) until it becomes clear using the video feed. This adjustment is best done with a ruler or some other writing/tool with clear indications of how sharp the the image is. Usually we do it in a lab/shed/workshop so it's easier to see everything.

A focused image helps a lot with detecting the weeds - an unfocused camera will mean smaller weeds will be blurred into the background and the colours won't be as distinct.

manninb, 2022-10-09:
Ta, muchly.

Bill Manning Land Services - Advisory Services (Cropping)

North West Local Land Services

PO Box 546 |35-37 Abbott Street | Gunnedah NSW 2380

T: +61 2 6742 9210 | F:+61 2 6742 4022

M:  0428 607 731

***@***.******@***.***>

W: northwest.lls.nsw.gov.au<http://northwest.lls.nsw.gov.au/>

www.facebook.com/northwestlocallandservices<http://www.facebook.com/northwestlocallandservices>

closed 2022-10-10 by geezacoleman