geezacoleman/OpenWeedLocator #58: focusing issue

manninb, 2022-10-24:
Had an issue focusing after many successful attempts. I updated software  first but then had trouble getting the PID number, see below.

 pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ mv owl owl-old
pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator
Cloning into 'OpenWeedLocator'...
remote: Enumerating objects: 366, done.
remote: Counting objects: 100% (247/247), done.
remote: Compressing objects: 100% (131/131), done.
remote: Total 366 (delta 163), reused 160 (delta 114), pack-reused 119
Receiving objects: 100% (366/366), 2.87 MiB | 5.26 MiB/s, done.
Resolving deltas: 100% (187/187), done.
pi@raspberrypi:~ $ mv OpenWeedLocator owl
pi@raspberrypi:~ $ cd ~/owl
pi@raspberrypi:~/owl $ pip install -r requirements.txt
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.21.6)
Requirement already satisfied: imutils in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /usr/lib/python3/dist-packages (from -r requirements.txt (line 5)) (1.6.2)
Requirement already satisfied: pandas in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.3.5)
Requirement already satisfied: glob2 in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7)
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (from -r requirements.txt (line 8)) (0.7.0)
Requirement already satisfied: picamera in /usr/lib/python3/dist-packages (from -r requirements.txt (line 9)) (1.13)
Requirement already satisfied: pytz>=2017.3 in /home/pi/.local/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2022.5)
Requirement already satisfied: python-dateutil>=2.7.3 in /home/pi/.local/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2.8.2)
Requirement already satisfied: six>=1.5 in /usr/lib/python3/dist-packages (from python-dateutil>=2.7.3->pandas->-r requirements.txt (line 6)) (1.12.0)
WARNING: You are using pip version 21.1.2; however, version 22.3 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
pi@raspberrypi:~/owl $ chmod a+x greenonbrown.py
pi@raspberrypi:~/owl $ chmod a+x owl_boot.sh
pi@raspberrypi:~/owl $ cd ..
pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
pi@raspberrypi:~ $

geezacoleman, 2022-10-24:
The `ps -C greenonbrown.py` command will just list anything that is running with 'greenonbrown.py' in the name, so here it looks like `greenonbrown.py` isn't running so you can continue with the focusing by running `./greenonbrown.py -- show-display`. A bit more detail below on why we need to do this.

I also noticed that you need activate the `owl` virtual environment, before installing the `requirements.txt` file. At the moment, those packages have been installed outside of the virtual environment so they won't be accessed when `greenonbrown.py` runs. To fix that run:
```
pi@raspberrypi:~ $ cd ~/owl 
pi@raspberrypi:/owl $ workon owl
(owl) pi@raspberrypi:/owl $ pip install -r requirements.txt
```

Before running the pip install command, make sure you see `(owl)` at the start of the line. This tells you that you're in the owl virtual environment. 

For a bit more detail on the `ps -C`, the reason we do this is that when `greenonbrown.py` runs on startup it restricts access to the camera to only one program at once. Because it operates in the background, you can't just use Terminal to stop it. So if we want to rerun it, we need to make sure it isn't running in the background by finding its program ID (PID) with the `ps -C` command and then using `sudo kill PID`.

In your case, it looks like there isn't a program called `greenonbrown,py` running, so it won't appear when you enter that command. This means you can just continue with the setup, but definitely need to fix the installation of `requirements.txt` into the right virtual environment.

geezacoleman, 2022-10-24:
I realised this wasn't particularly clear in the updating instructions and I think it was resulting in both @manninb and @MerbeinRobotics issues with access to the RPi.GPIO package. 

This has hopefully been made clearer in this commit: https://github.com/geezacoleman/OpenWeedLocator/commit/d62af8c45de26da92276d695c37f9982cb262f0f

manninb, 2022-10-25:
Got this sorted. Thanks for your help.

closed 2022-10-25 by manninb