geezacoleman/OpenWeedLocator #68: moving to 50 cm nozzle.

manninb, 2023-03-01:
What would be needed to operate the systems for a 50cm nozzle spacing? Does anybody have the capacity to do the coding if required. Most commercial spraying works on 50 cm spacing.

geezacoleman, 2023-03-01:
There are a couple of ways to change this:

1. You could raise the height of the camera to 2m so that four lanes cover 50cm each, but I don't think the resolution will be high enough for good performance on small weeds.
2. Best option is to change `nozzleNum=4` to `nozzleNum=2` and leave the OWL at 1m height. The OWL will only use the first two values in the dictionary that maps the nozzle to the pin on the Raspberry Pi that turns it on. So just make sure you connect the two solenoids to those pins.

```
      # nozzleDict maps the reference nozzle number to a boardpin on the embedded device
        self.nozzleDict = {
            0: 13,         <- it will only use these first two pins
            1: 15,         <- and this one
            2: 16,
            3: 18
        }
```

The `nozzleNum` value is found in `greenonbrown.py` [Line 605](https://github.com/geezacoleman/OpenWeedLocator/blob/main/greenonbrown.py#L605):

```
    owl = Owl(videoFile=args.video_file,
              show_display=args.show_display,
              recording=args.recording,
              exgMin=25,
              exgMax=200,
              hueMin=39,
              hueMax=83,
              saturationMin=50,
              saturationMax=220,
              brightnessMin=60,
              brightnessMax=190,
              resolution=(416, 320),
              nozzleNum=2,                                   <-- this is what you want to change from 4 to 2 (or however many you like)
              framerate=args.framerate,
              exp_mode=args.exp_mode,
              exp_compensation=args.exp_compensation,
              awb_mode=args.awb_mode,
              sensor_mode=args.sensor_mode,
              parameters_json=None
              )

```

geezacoleman, 2023-03-06:
Keep me posted about how this goes @manninb! Would be great to get some videos/photos of the 50cm nozzle setup. I'll close it for now, but feel free to reopen/add more comments if needed.

closed 2023-03-06 by geezacoleman

geezacoleman, 2023-12-02:
That sounds like an interesting use case, would be great if you can share some photos of the completed setup over in the community builds page once you're done!

What specific issues were you having with changing nozzle number?

If it's just one nozzle, change the `nozzleNum=4` parameter on [Line 528](https://github.com/geezacoleman/OpenWeedLocator/blob/be7b979907b8f94b745fa0090c3ff126017f3f3a/owl.py#L528C27-L528C27) in `owl.py` to 1. That way the image will be divided into a single zone and only activate one nozzle. 

The `nozzleDict` I talk about above just maps each zone within the image, to a GPIO pin on the Pi. These pins are connected to the relay control board. So if the wrong GPIO number is used because the mapping is incorrect, then the wrong pin is activated and the relay control board probably won't activate, or perhaps the wrong section is activated. As mentioned above the system will only use the first nozzle in the dictionary, but to be certain, you can delete the other parts of the dictionary so it will be 

```python
self.nozzleDict = {
            0: 13        # it will only use this pin
        }
```

Just make sure the wire from GPIO 13 is connected. You can copy this code in exactly as it is here to replace the dictionary starting [Line 114 in `owl.py`](https://github.com/geezacoleman/OpenWeedLocator/blob/be7b979907b8f94b745fa0090c3ff126017f3f3a/owl.py#L114-L120)  if that helps.

I've tested this on my own laptop (not OWL device though) and it seems to work, but let me know how you go and if you have any more questions.

Kmrfarmsinc, 2023-12-02:
So, I was able to find the owl.py doc to change the number of nozzles to 1 and the self NozzleDict to only read 0: 13, but when I try and run it the program says 0 nozzles when starting up through the terminal. Also, I had the camera working but it randomly shut off after about 20 seconds from running the program. Don’t know if I deleted more than I should or what but these are the pictures of the terminal and the owl.py changes. 
[image]
[image]
[image]

geezacoleman, 2023-12-02:
In Python things are generally 0 indexed, so they start counting from 0 not 1. I carried this over into the nozzle numbering on the display (hence Nozzle 0) appears, but that just means the first nozzle, not that there are zero nozzles operating. I think I'll update that for the label above each nozzle, because it isn't entirely clear. This means it is working correctly.

I would recommend running `./owl.py --show-display --exp-mode 'auto'` and see if it closes after 20s or not. I think in the focusing mode it may be running into issues with the almost entirely dark image and the division by 0. In your dictionary you do have an unnecessary comma after 13. So I would delete that too and try again.

There is potentially an unseen bug there somewhere too, but hard to say what just yet!

geezacoleman, 2023-12-04:
Improvements to some of these issues have been made in #103.

Kmrfarmsinc, 2023-12-11:
Having a problem again with the camera feed cutting out. I was able to get the PID, kill that program, get the camera to pop up but it will turn off after 10-20 seconds. Any solution?

geezacoleman, 2023-12-11:
A few questions, which should hopefully narrow it down:
* Similar to above, are you able to provide a screenshot of how you are running it in Terminal and when it fails? 
* Are you using just the default green-on-brown detection mode? 
* Have you checked the power cables and camera cable to make sure they are connected securely? I would recommend power it over the GPIO pins instead of the USB-C. The USB-C has caused me to have really difficult to diagnose issues before.
* Could you include an image of the hardware too?
* Did you start with the OWL image file and have you updated it recently with `update_owl.sh`?

Kmrfarmsinc, 2023-12-11:
Yes I am using the default green-on-brown detection mode. 

The power cord are all good, tested those myself.

I see the GPIO pins but I have the expansion board that goes in those pins. I am going to include the pictures of the hardware. 

The camera will turn on and remain on. The problem I found is that I turned my phone camera on and was moving it closer and further away to try and get a detection. I got the camera to pick it up but when i get about 20 cm away from the lens, its shut down. It makes me think it is a sensitivity issue but im not sure. 

Im using a HD Raspberry Pi with the ribbon cable.
[image]
[image]

geezacoleman, 2023-12-11:
Ok thanks for sharing the hardware setup - as you say, power looks a lot more robust than the USB-C adapter, so I don't think that should be an issue. When you say you move the phone camera closer/further away, is that like a green object or similar to test it? If you're moving the Pi HD camera and it cuts out, I'm guess that might be a cable issue then. In either case, I'd double check the camera ribbon connection. I have had similar issues previously with old cables that have been bent too much.

For the software, do you have the latest version of the OWL software? I ask because we had this issue from some buggy code a while back but have since fixed it. To diagnose the software - can you leave it running with no detections being made? There might be an issue running only one detection zone and detections falling outside that.

There shouldn't be an issue getting too close to the camera - I've done that plenty of times, it will just get a bit blurry but won't cause the camera to cut out. I would advise against running focus_owl.sh too long though - there could be issues there with too much blur.

To summarise:
1. check ribbon cable
2. leave software running with/without green detections and make sure it is updated using update_owl.sh.
2a. try running it with the default four lanes and see if it works. If it does but fails with 1 lane/detection zone then you'll know it's that.
3. Make sure it isn't running the focusing script. 
4. are you using the original owl image downloaded form github or the latest Raspbian OS from Raspberry Pi? The latter will cause issues because it will want to run libcamera not picamera.

Kmrfarmsinc, 2023-12-15:
The ribbon for the camera was a little loose but it still happened after I fixed the ribbon. 

I am going to give a picture below and I’m wondering if this could be the problem. 
[image]

geezacoleman, 2023-12-16:
In that screenshot, you're working in the global environment for the OWL - we run the OWL software in a virtual environment, which is a way of keeping all the various other packages together. Helps with version control and managing conflicts. The issue is that anything outside the virtual environment `owl` is inaccessible to the owl software.

If you pip install things. always pip install them into the virtual environment `owl`, otherwise you'll run into issues.

To install into the virtual environment, you first need to activate it and then run the pip install commands.

```
workon owl
pip install --upgrade pip
pip install blessed
```

But I would just recommend running:

```
workon owl
bash update_owl.sh
```

This will download the latest OWL software, and install all the right dependencies into the right virtual environment too.