geezacoleman/OpenWeedLocator #76: green on green

manninb, 2023-05-04:
Hi All,
I have read some comments on the site about green on green. I'm just wondering if (for the technically incompetent)  you could give me a summary of where that is up to and the potential to update my existing unit to green on green? I understand the large time commitment in training such a system through  the bit of work I did with autoweed.

geezacoleman, 2023-05-10:
Hi Bill,

Overall I'd describe it as a beta version - it works on my desktop/Pi but it hasn't been field tested yet. Largely due to lack of datasets available to train models for it and fields to use it in. Hoping to change that this season.

To summarise, we developed the green-on-green capability using the Google Coral, which is a retrofit device for the Raspberry Pi to give it the power needed to run these large models. You just need to connect it via USB to the Pi, install a few things [as per the instructions here](https://github.com/geezacoleman/OpenWeedLocator/tree/main/models) and you're ready to run your own models. If you had a model in the correct format, you would drag it to the models directory and the system does the rest.

As you've identified, training the model is a big hurdle. But there are many high performing architectures (e.g. YOLOv5/v8/NAS) that are easy to get relatively high results from on smaller datasets that are good for prototypes.

So in short - very much possible to update your unit to green-on-green using the guide above, though it is still an early version. Any field testing you do would be greatly appreciated to iron out some issues. The challenge will be collecting data and training your own model for this purpose.

manninb, 2023-05-12:
Is the YOLO only for training ie, is it loaded on to the spray units or does it develop an algorithm that is used with the software you have written?

geezacoleman, 2023-05-12:
YOLO is an algorithm architecture (think of it like an unfurnished house) - if you give it weeds data to train on you can then load that trained/furnished algorithm onto the OWL to use to detect weeds. YOLOv5/v8 can be trained on Google Colab, an online tool for coding which gives you access to a high power graphics card for free. 

The issues you're having with owl.py/greenonbrown.py are the result of trying to incorporate the outputs from the YOLO/other algorithms too (so my apologies for the hassle!).

manninb, 2023-06-14:
Hi 
I'm interested in collecting images to see if I could get a green on green version going. In short I would be happy to put time in on this but I would need support in setting up the training software and incorporating the results into the OWL. This support I would think have to come from Syd Uni, so my question would be can you supply that?

geezacoleman, 2023-06-20:
Hi Bill, flick me an email and we can organise this further - would need to work out how many images/target weeds etc. but sounds like an interesting proposal

manninb, 2023-06-21:
Hi Guy,
My initial thought is could I use the OWL to collect the images, then  load them into some software and then go through and ID them? I don't know if this is practical,  storage  on the respberry may be an issue.  I think that is what the Auto Weed did when I was working with them. They were able to do a fairly fast turnaround  in terms of capturing images and Identifying them and training the model. In terms of how many  images are needed I couldn't say.  I would need  substantial help on the software side of things, but I could put in a fair chunk of time  capturing classifying images and testing the unit.  As a  research organisation Sydney uni may feel this is more a development job, not pure research and therefore not within the remit of the Uni.  Do we need to have a chat on the phone or teams etc?

I'm not sure which email for you is best to use.

Bill Manning Land Services - Advisory Services (Cropping)

North West Local Land Services

PO Box 546 |35-37 Abbott Street | Gunnedah NSW 2380

T: +61 2 6742 9210 | F:+61 2 6742 4022

M:  0428 607 731

***@***.******@***.***>

W: northwest.lls.nsw.gov.au<http://northwest.lls.nsw.gov.au/>

www.facebook.com/northwestlocallandservices<http://www.facebook.com/northwestlocallandservices>

manninb, 2023-06-21:
Also I should have mentioned that I'm doing a talk at the weeds conference in Dubbo in August  on the work I have done with the OWL.  If you wish to have any input I'm more than happy to accommodate that.

Bill Manning Land Services - Advisory Services (Cropping)

North West Local Land Services

PO Box 546 |35-37 Abbott Street | Gunnedah NSW 2380

T: +61 2 6742 9210 | F:+61 2 6742 4022

M:  0428 607 731

***@***.******@***.***>

W: northwest.lls.nsw.gov.au<http://northwest.lls.nsw.gov.au/>

www.facebook.com/northwestlocallandservices<http://www.facebook.com/northwestlocallandservices>

closed 2024-01-03 by geezacoleman