geezacoleman/OpenWeedLocator #78: Focusing camera (again)

manninb, 2023-05-11:
Sorry for the hassle but having not done this for a while, I'm again having an issue getting the camera view up.

I opened up the "terminal" from the main raspberry screen and did the following (see below). somehow I have come at it from the wrong wrong point but cant seem to find the correct one.

 I keeping getting this on the end of the command line  "/owl $"

pi@raspberrypi:~ $ cd ~/owl
pi@raspberrypi:~/owl $ workon owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.21.0)
Requirement already satisfied: imutils in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Requirement already satisfied: pandas in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.2.5)
Requirement already satisfied: glob2 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7)
Requirement already satisfied: RPi.GPIO in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 8)) (0.7.1)
Requirement already satisfied: picamera in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 9)) (1.13)
Requirement already satisfied: colorzero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from gpiozero->-r requirements.txt (line 5)) (2.0)
Requirement already satisfied: pytz>=2017.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2021.1)
Requirement already satisfied: python-dateutil>=2.7.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2.8.1)
Requirement already satisfied: six>=1.5 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from python-dateutil>=2.7.3->pandas->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: setuptools in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from colorzero->gpiozero->-r requirements.txt (line 5)) (57.0.0)
WARNING: You are using pip version 21.1.2; however, version 23.1.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~/owl $ ps -C owl.py
  PID TTY          TIME CMD
(owl) pi@raspberrypi:~/owl $

geezacoleman, 2023-05-11:
No worries - based on what you've copied in, it looks as though all the commands you have entered have worked, the final `ps -C owl.py` command will check if `owl.py` is running. If it returns just the headings 'PID TTY TIME CMD', then the program isn't running in the background. Is that what you mean? The focusing camera method is quite annoying and doesn't need to be so fiddly, so I've started a new issue #79 to resolve some of these hurdles. Ideally the code should just check everything itself.

In the meantime, if you want to run the software, try typing `./owl.py` or `python owl.py` to start the program. If you want to see the display make sure to add the `--show-display` flag too.

Hopefully this answers the question - feel free to post up more of the output if I have missed something and I can try debugging it for you.

manninb, 2023-05-11:
Hi. Still having no luck. I think there is something more fundamental I'm missing. I'll try another unit and see if I have any luck.

pi@raspberrypi:~ $ ./owl.py
bash: ./owl.py: No such file or directory
pi@raspberrypi:~ $ python owl.py 
python: can't open file 'owl.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ ps -C owl.py
  PID TTY          TIME CMD
pi@raspberrypi:~ $ python owl.py 
python: can't open file 'owl.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $ ./owl.py
bash: ./owl.py: No such file or directory
pi@raspberrypi:~ $ ./owl.py

manninb, 2023-05-11:
tried on another unit and got a little further. (see below).

pi@raspberrypi:~ $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  542 ?        00:12:42 greenonbrown.py
pi@raspberrypi:~ $ cd ~/owl
pi@raspberrypi:~/owl $ workon owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.21.0)
Requirement already satisfied: imutils in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Requirement already satisfied: pandas in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.2.5)
Requirement already satisfied: glob2 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7)
Requirement already satisfied: RPi.GPIO in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 8)) (0.7.1)
Requirement already satisfied: picamera in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 9)) (1.13)
Requirement already satisfied: colorzero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from gpiozero->-r requirements.txt (line 5)) (2.0)
Requirement already satisfied: pytz>=2017.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2021.1)
Requirement already satisfied: python-dateutil>=2.7.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2.8.1)
Requirement already satisfied: six>=1.5 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from python-dateutil>=2.7.3->pandas->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: setuptools in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from colorzero->gpiozero->-r requirements.txt (line 5)) (57.0.0)
WARNING: You are using pip version 21.1.2; however, version 23.1.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~/owl $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  542 ?        00:22:25 greenonbrown.py
(owl) pi@raspberrypi:~/owl $ sudo kill 542
(owl) pi@raspberrypi:~/owl $ ~/owl/./owl.py --show-display
bash: /home/pi/owl/./owl.py: No such file or directory
(owl) pi@raspberrypi:~/owl $ sudo kill542
sudo: kill542: command not found
(owl) pi@raspberrypi:~/owl $ sudo kill 542
kill: (542): No such process
(owl) pi@raspberrypi:~/owl $

manninb, 2023-05-11:
I seemed to have turned this unit off and cant turn it back on

pi@raspberrypi:~ $ ps -C owl.py
 
PID TTY          TIME CMD
pi@raspberrypi:~ $ ./owl.py
bash: ./owl.py: No such file or directory
pi@raspberrypi:~ $

manninb, 2023-05-15:
when updating got an error message so I reflashed the card and then updated (see below) still got an error message managed to get the PID but still cant get camera up. Would it be possible to talk on the phone?, it might save you some time.

pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ cd ~
(owl) pi@raspberrypi:~ $ mv owl owl-old
(owl) pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator
Cloning into 'OpenWeedLocator'...
remote: Enumerating objects: 535, done.
remote: Counting objects: 100% (183/183), done.
remote: Compressing objects: 100% (65/65), done.
remote: Total 535 (delta 156), reused 118 (delta 118), pack-reused 352
Receiving objects: 100% (535/535), 23.36 MiB | 4.67 MiB/s, done.
Resolving deltas: 100% (285/285), done.
(owl) pi@raspberrypi:~ $ mv OpenWeedLocator owl
(owl) pi@raspberrypi:~ $ cd ~/owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy~=1.21.2
  Downloading numpy-1.21.6.zip (10.3 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 10.3 MB 65 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: imutils~=0.5.4 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Collecting pandas~=1.3.2
  Downloading https://www.piwheels.org/simple/pandas/pandas-1.3.5-cp37-cp37m-linux_armv7l.whl (14.4 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 14.4 MB 17 kB/s 
Requirement already satisfied: glob2 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7)
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp37-cp37m-linux_armv7l.whl (47 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 47 kB 1.4 MB/s 
Requirement already satisfied: picamera in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 9)) (1.13)
Collecting piexif~=1.1.3
  Downloading https://www.piwheels.org/simple/piexif/piexif-1.1.3-py2.py3-none-any.whl (20 kB)
Collecting folium~=0.14.0
  Downloading https://www.piwheels.org/simple/folium/folium-0.14.0-py2.py3-none-any.whl (102 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 102 kB 83 kB/s 
Collecting opencv-python~=4.5.5.64
  Downloading https://www.piwheels.org/simple/opencv-python/opencv_python-4.5.5.64-cp37-cp37m-linux_armv7l.whl (11.1 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 11.1 MB 2.9 kB/s 
ERROR: Could not find a version that satisfies the requirement pycoral~=2.0.0 (from versions: 0.0.4, 0.0.5, 0.0.6, 0.0.7, 0.0.8, 0.0.9, 0.1.0)
ERROR: No matching distribution found for pycoral~=2.0.0
WARNING: You are using pip version 21.1.2; however, version 23.1.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~/owl $ workon owl
(owl) pi@raspberrypi:~/owl $ ps -C greenonbrown.py
  PID TTY          TIME CMD
  534 ?        00:55:38 greenonbrown.py
(owl) pi@raspberrypi:~/owl $ sudo kill 534
(owl) pi@raspberrypi:~/owl $ ~/owl/./owl.py --show-display
bash: /home/pi/owl/./owl.py: Permission denied
(owl) pi@raspberrypi:~/owl $ 

Then I tried again

owl) pi@raspberrypi:~/owl $ cd ~/owl
(owl) pi@raspberrypi:~/owl $ ./greenonbrown.py --show-displa
bash: ./greenonbrown.py: Permission denied
(owl) pi@raspberrypi:~/owl $

geezacoleman, 2023-05-15:
Thanks for persisting! I reckon you should almost be there - in the updating procedure, you just need to run the final two steps to change the permissions on the new files you just downloaded.

Run these lines and then it should hopefully work:
```
(owl) pi@raspberrypi:~/owl $ chmod a+x owl.py
(owl) pi@raspberrypi:~/owl $ chmod a+x owl_boot.sh
```

Then you can run:
```
./owl.py --show-display
```

If you don't have any luck after this - happy to catch up over the phone and try getting it working for you. Just send me an email and we can schedule a time.

manninb, 2023-05-16:
Still no luck  see below, I could talk today (weds) or tomorrow or early next week, I'm out Thursday Friday. Is there some type of zoom app we use on the raspberry, so you can see what's happening this end?

pi@raspberrypi:~ $ chmod a+x owl.py
chmod: cannot access 'owl.py': No such file or directory
pi@raspberrypi:~ $ chmod a+x owl_boot.sh
chmod: cannot access 'owl_boot.sh': No such file or directory
pi@raspberrypi:~ $ ./owl.py --show-display
bash: ./owl.py: No such file or directory
pi@raspberrypi:~ $

manninb, 2023-05-16:
I reflashed the card again, updated the software (no errors this time), managed to get the PID but still couldn't get the camera up.

pi@raspberrypi:~ $ cd ~/owl
pi@raspberrypi:~/owl $ workon owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.21.0)
Requirement already satisfied: imutils in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Requirement already satisfied: pandas in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.2.5)
Requirement already satisfied: glob2 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7)
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp37-cp37m-linux_armv7l.whl (47 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 47 kB 115 kB/s 
Requirement already satisfied: colorzero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from gpiozero->-r requirements.txt (line 5)) (2.0)
Requirement already satisfied: pytz>=2017.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2021.1)
Requirement already satisfied: python-dateutil>=2.7.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2.8.1)
Requirement already satisfied: six>=1.5 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from python-dateutil>=2.7.3->pandas->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: setuptools in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from colorzero->gpiozero->-r requirements.txt (line 5)) (57.0.0)
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.1
WARNING: You are using pip version 21.1.2; however, version 23.1.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~/owl $ ps -C owl.py
  PID TTY          TIME CMD
(owl) pi@raspberrypi:~/owl $ ./owl.py
bash: ./owl.py: No such file or directory
(owl) pi@raspberrypi:~/owl $ python owl.py
python: can't open file 'owl.py': [Errno 2] No such file or directory
(owl) pi@raspberrypi:~/owl $ cd ~
(owl) pi@raspberrypi:~ $ mv owl owl-old
(owl) pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator
Cloning into 'OpenWeedLocator'...
remote: Enumerating objects: 539, done.
remote: Counting objects: 100% (187/187), done.
remote: Compressing objects: 100% (69/69), done.
remote: Total 539 (delta 157), reused 118 (delta 118), pack-reused 352
Receiving objects: 100% (539/539), 23.36 MiB | 3.35 MiB/s, done.
Resolving deltas: 100% (286/286), done.
(owl) pi@raspberrypi:~ $ mv OpenWeedLocator owl
(owl) pi@raspberrypi:~ $ cd ~/owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy~=1.21.2
  Downloading numpy-1.21.6.zip (10.3 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 10.3 MB 65 kB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: imutils~=0.5.4 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Collecting pandas~=1.3.2
  Downloading https://www.piwheels.org/simple/pandas/pandas-1.3.5-cp37-cp37m-linux_armv7l.whl (14.4 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 14.4 MB 17 kB/s 
Requirement already satisfied: RPi.GPIO in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 7)) (0.7.1)
Requirement already satisfied: picamera in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 8)) (1.13)
Collecting opencv-python~=4.5.5.64
  Downloading https://www.piwheels.org/simple/opencv-python/opencv_python-4.5.5.64-cp37-cp37m-linux_armv7l.whl (11.1 MB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 11.1 MB 2.9 kB/s 
Collecting tqdm~=4.64.1
  Downloading https://www.piwheels.org/simple/tqdm/tqdm-4.64.1-py2.py3-none-any.whl (78 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 78 kB 204 kB/s 
[... cut ...]
    from image_sampler import bounding_box_image_sample, square_image_sample, whole_image_save
  File "/home/pi/owl/image_sampler.py", line 3, in <module>
    import cv2
  File "/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/cv2/__init__.py", line 9, in <module>
    from .cv2 import _registerMatType
ImportError: cannot import name '_registerMatType' from 'cv2.cv2' (/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so)
(owl) pi@raspberrypi:~/owl $

manninb, 2023-05-16:
My email  is william.manning@lls.nsw.gov.au

geezacoleman, 2023-05-17:
Thanks Bill - I sent an email yesterday afternoon, will just go through these errors here too.

Good news is `owl.py` is now running but it's running into an issue with the installation of opencv. This problem is caused by my PyCharm automatically updating the requirements.txt file. I've fixed that - so if you reflash the card again and then update the owl directories and run `pip install requirements.txt` as before it should work.

```
Traceback (most recent call last):
File "/home/pi/owl/./owl.py", line 5, in
from image_sampler import bounding_box_image_sample, square_image_sample, whole_image_save
File "/home/pi/owl/image_sampler.py", line 3, in
import cv2
File "/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/cv2/init.py", line 9, in
from .cv2 import _registerMatType
ImportError: cannot import name '_registerMatType' from 'cv2.cv2' (/home/pi/.virtualenvs/owl/lib/python3.7/site-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so)
```

Just double check the `requirements.txt` file has:
```
numpy
imutils
gpiozero
pandas
RPi.GPIO
picamera
tqdm
```

Though most should already be installed.

manninb, 2023-05-22:
Reflashed card, updated owl  and got the the correct text in the requirements file. I  Could not get a PID (I noticed the OWL stopped running at some point  during the process). I could not get the OWL running after looking at your previous instructions. Could you explain again how to get the OWL running again, perhaps with a little more detail?

pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ cd ~
(owl) pi@raspberrypi:~ $ mv owl owl-old
(owl) pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator
Cloning into 'OpenWeedLocator'...
remote: Enumerating objects: 552, done.
remote: Counting objects: 100% (200/200), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 552 (delta 162), reused 118 (delta 118), pack-reused 352
Receiving objects: 100% (552/552), 23.37 MiB | 2.80 MiB/s, done.
Resolving deltas: 100% (291/291), done.
(owl) pi@raspberrypi:~ $ mv OpenWeedLocator owl
(owl) pi@raspberrypi:~ $ cd ~/owl
(owl) pi@raspberrypi:~/owl $ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: numpy in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 3)) (1.21.0)
Requirement already satisfied: imutils in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 4)) (0.5.4)
Requirement already satisfied: gpiozero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.6.2)
Requirement already satisfied: pandas in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.2.5)
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp37-cp37m-linux_armv7l.whl (47 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 47 kB 112 kB/s
Requirement already satisfied: picamera in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from -r requirements.txt (line 8)) (1.13)
Collecting tqdm
  Downloading https://www.piwheels.org/simple/tqdm/tqdm-4.65.0-py3-none-any.whl (77 kB)
     |\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588\u2588| 77 kB 96 kB/s
Requirement already satisfied: colorzero in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from gpiozero->-r requirements.txt (line 5)) (2.0)
Requirement already satisfied: python-dateutil>=2.7.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2.8.1)
Requirement already satisfied: pytz>=2017.3 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from pandas->-r requirements.txt (line 6)) (2021.1)
Requirement already satisfied: six>=1.5 in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from python-dateutil>=2.7.3->pandas->-r requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: setuptools in /home/pi/.virtualenvs/owl/lib/python3.7/site-packages (from colorzero->gpiozero->-r requirements.txt (line 5)) (57.0.0)
Installing collected packages: tqdm, RPi.GPIO
Successfully installed RPi.GPIO-0.7.1 tqdm-4.65.0
WARNING: You are using pip version 21.1.2; however, version 23.1.2 is available.
You should consider upgrading via the '/home/pi/.virtualenvs/owl/bin/python -m pip install --upgrade pip' command.
(owl) pi@raspberrypi:~/owl $ chmod a+x owl.py
(owl) pi@raspberrypi:~/owl $ chmod a+x owl_boot.sh
(owl) pi@raspberrypi:~/owl $ ps -C owl.py
  PID TTY          TIME CMD
(owl) pi@raspberrypi:~/owl $ greenonbrown.py:
bash: greenonbrown.py:: command not found
(owl) pi@raspberrypi:~/owl $ cd ~/owl
(owl) pi@raspberrypi:~/owl $ ./greenonbrown.py --show-display
bash: ./greenonbrown.py: Permission denied
(owl) pi@raspberrypi:~/owl $

geezacoleman, 2023-05-27:
I've spent some time putting together/testing a few different scripts (#89 and #86 ) to somewhat automate both focusing and updating process. This should hopefully avoid a lot of the hassle.

To use both of these, you'll need to clone the latest repository. It assumes you have the old image flashed to the SD card. 

In a new terminal window:

## Step 1 - Update the OWL
```
pi@raspberrypi: $ cd ~                  # makes sure you are in the home directory
pi@raspberrypi: $ mv owl owl-old            # renames owl directory to owl-old
pi@raspberrypi: $ git clone https://github.com/geezacoleman/OpenWeedLocator        # 
pi@raspberrypi: $ workon owl                             # activates the owl environment (very important for running the update script)
(owl) pi@raspberrypi: $ cd owl                   # moves into the owl directory that you just downloaded
(owl) pi@raspberrypi:/owl $ bash update_owl.sh
```
Once you run the final `update_owl.sh` command it will:
1. update the whole system
2. download the new owl software again
3. install the requirements
4. change all the permissions of the relevant files automatically. 

I've tested this a few times on my own system and it works fine. Just make sure you are in the `owl` virtual environment otherwise the packages won't be installed into the correct area.

During this process you will be asked to confirm a few things - just type `y` and press enter to continue. It will take about 5 - 10 mins to download and install everything given the age of the image. Do not interrupt it or it will brick the SD card (found this out the hard way accidentally).

Assuming it all goes to plan, you should have everything in the right place and with the right permissions after that has run.

## Step 2 - focus the OWL
I have also automated this process somewhat and added the value for blurriness/clarity to the display image. Once you have updated the OWL, you just need to run:

```
(owl) pi@raspberrypi:/owl $ bash focus_owl.sh
```

This will close any operating `greenonbrown.py` / `owl.py` instances and start the software with the --show-display and --focus flags. You should then be able to sort this out.

Here's an example of the new focusing layout. Blurry image has a low value while the clear image has a high value - moving the focusing ring on the lens to maximise this value. It's not perfect so double check it looks focused to you as well.
| Blurry Image | Clear Image |
|--------------|-------------|
|[image] | [image] |

manninb, 2023-05-30:
That seems to be working - thanks very much, sorry for all the  back and forth.

geezacoleman, 2023-05-30:
Glad to hear it worked out for you! Thanks for helping improve this whole procedure and finding out some bugs along the way. I'll close this now as completed.

closed 2023-05-30 by geezacoleman