geezacoleman/OpenWeedLocator #83: Yikes! That's an expensive voltage regulator!

Minorplanets76, 2023-05-20:
The 5V regulator specified to power the Pi is as expensive as the Camera! I'm sure it's very good, but is it necessary?  Should other solutions be considered, e.g. POE for Ethernet connected modules or a big regulator powering multiple Pi's?

plains203, 2023-05-20:
That's the beauty of an open system. You can power it however you like.

geezacoleman, 2023-05-21:
The Pololu ones certainly [aren't cheap](https://www.pololu.com/product/2850), but given how noisy some of the incoming 12V might be it's nice to invest in a good quality power supply. They accept a wide range of input voltages and provide 5A output in case you need to run any other 5V things out of the OWL. There is also the 2.5A option which is a fair amount cheaper.

Lots of different 12V to 5V options out there that aren't pololu and are probably also quite good. If you're interested in having some of your own things built - check out this [driver board](https://github.com/geezacoleman/owl-driver-board) my brother built for the OWL. It incorporates the buck converter and the solid state relays into one board. Has PWM capabilities and is tolerant to noise/reverse polarity etc. Would recommend - having it built yourself can be cheap on a per unit basis.

Minorplanets76, 2023-05-21:
That's perfect. I'll build some of those. Already all over it. Good stuff.

closed 2023-05-21 by Minorplanets76