geezacoleman/OpenWeedLocator #91: Focussing

manninb, 2023-06-14:
Had a problem with the new focusing procedure when I went to my other unit. They had old software on them but when running through the procedure I got the following errors.
pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ mv owl owl-old
pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator
Cloning into 'OpenWeedLocator'...
remote: Enumerating objects: 599, done.
remote: Counting objects: 100% (247/247), done.
remote: Compressing objects: 100% (125/125), done.
remote: Total 599 (delta 184), reused 132 (delta 122), pack-reused 352
Receiving objects: 100% (599/599), 23.38 MiB | 4.05 MiB/s, done.
Resolving deltas: 100% (313/313), done.
pi@raspberrypi:~ $ workon owl
(owl) pi@raspberrypi:~ $ cd owl
bash: cd: owl: No such file or directory
(owl) pi@raspberrypi:~ $ bash update_owl.sh
bash: update_owl.sh: No such file or directory
(owl) pi@raspberrypi:~ $ 

I tried again after reflashing the cards but got the same error.
Any assistance greatly appreciated.

geezacoleman, 2023-06-14:
Looks like when you clone this repository it isn't being renamed to `owl` and is instead just staying as OpenWeedLocator. To fix this, run everything above again, but change this line and add `owl` after the URL:

```
pi@raspberrypi:~ $ git clone https://github.com/geezacoleman/OpenWeedLocator owl
```

Alternatively just open up a file browser and go to the Home folder and change the name of the OpenWeedLocator directory to owl. That should fix it.

geezacoleman, 2023-06-20:
Will close this one now - let me know if you've had any other issues.

closed 2023-06-20 by geezacoleman