geezacoleman/OpenWeedLocator #100: trigger delay

manninb, 2023-11-03:
Is there a way to shorten time between detection and nozzle activation as my unit seems to be triggering a bit late even at slow speeds.
I see a number there "delay" can that be shortened to less than zero?

geezacoleman, 2023-11-03:
Hi Bill,

It can't be shortened more than 0 unfortunately, so there's no easy software fix for this. 

There are a few areas that can cause this though and can be fixed to improve performance:
1. Which software version are you using?
the original software is slower than the current version, so make sure it is updated.

2. check image resolution - should be set to 416 x 320.
High resolution will slow detection speed on the Pi

3. how far in front of the nozzles is the camera and/or is there a slight angle backwards? To fix this you can increase this distance or slightly angle the camera forwards. This is probably the simplest solution.

4. what sort of nozzles are you using?
I bought some cheap ones recently that are very slow to turn on and as a result will miss all weeds even at slow speeds.

I think improvements can be made in the efficiency of activating nozzles - it is quite a slow system currently. I'll add it to the todo list for the project.

manninb, 2023-11-06:
thanks for that I'll  check all the above. We are using goyen solenoids and nozzles, do you regard those as fast or slow? 
[image]

geezacoleman, 2024-01-03:
Hi Bill, I think I've found a potentially major source of delay that is contributing to your error. I'm working on a proper solution to this that lets people retain the logging of all detections, but for your case I think it's best to remove it for the moment.

To fix this:
In `relay_control.py` go to [Lines 142 - 143](https://github.com/geezacoleman/OpenWeedLocator/blob/962e3fbf6a9e7a1be822a516436cb5b7f8a3e2e0/relay_control.py#L142C4-L143) and delete them both.

````python
line = f"nozzle: {nozzle} | time: {timeStamp} | location {location} | delay: {delay} | duration: {duration}"
self.logger.log_line(line, verbose=False)
```

These are the two lines you want to remove. It will speed up the detection process a considerable amount. I tested it on my laptop and the time taken to run through each detection and activate it went from ~10ms to <0.01ms (effectively unrecordable on Python). So this will reduce the delay between the detection and actually activating the nozzle, hopefully going someway to solving your issue.

The only downside is the detection times and the information above will not be recorded. As mentioned above, I will be working on a longer term solution to this as a priority given how much of a performance increase it provides.

manninb, 2024-01-04:
Thanks again for your help, I’m away for a couple of weeks, but I’ll be back onto this soon.

Get Outlook for iOS<https://aka.ms/o0ukef>
________________________________
From: Guy Coleman ***@***.***>
Sent: Thursday, January 4, 2024 12:20:56 AM
To: geezacoleman/OpenWeedLocator ***@***.***>
Cc: Bill Manning ***@***.***>; Author ***@***.***>
Subject: Re: [geezacoleman/OpenWeedLocator] trigger delay (Issue #100)

Hi Bill, I think I've found a potentially major source of delay that is contributing to your error. I'm working on a proper solution to this that lets people retain the logging of all detections, but for your case I think it's best to remove it for the moment.

To fix this:
In relay_control.py go to Lines 142 - 143<https://github.com/geezacoleman/OpenWeedLocator/blob/962e3fbf6a9e7a1be822a516436cb5b7f8a3e2e0/relay_control.py#L142C4-L143> and delete them both.

line = f"nozzle: {nozzle} | time: {timeStamp} | location {location} | delay: {delay} | duration: {duration}"
self.logger.log_line(line, verbose=False)
```

These are the two lines you want to remove. It will speed up the detection process a considerable amount. I tested it on my laptop and the time taken to run through each detection and activate it went from ~10ms to <0.01ms (effectively unrecordable on Python). So this will reduce the delay between the detection and actually activating the nozzle, hopefully going someway to solving your issue.

The only downside is the detection times and the information above will not be recorded. As mentioned above, I will be working on a longer term solution to this as a priority given how much of a performance increase it provides.

—
Reply to this email directly, view it on GitHub<https://github.com/geezacoleman/OpenWeedLocator/issues/100#issuecomment-1875363878>, or unsubscribe<https://github.com/notifications/unsubscribe-auth/AN3U5AQTTVKOROSS7JEIA5DYMVLLRAVCNFSM6AAAAAA632KU2KVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQNZVGM3DGOBXHA>.
You are receiving this because you authored the thread.Message ID: ***@***.***>

geezacoleman, 2024-11-14:
Hi Bill - the latest software version should be a lot better with this. Would recommend updating. 

I'll still work on finding ways to decrease time between weed entering FOV and solenoid activating but for now I'll close this issue. Feel free to post any other comments/questions anytime though.