geezacoleman/OpenWeedLocator #104: model testing

zerosnsa, 2023-12-11:
May I ask why this error occurs when my model passes the specified algorithm。
[image]

geezacoleman, 2023-12-11:
This error is most likely because pycoral isn't installed correctly, but can be because of a number of issues with the `GreenonGreen` class. Unfortunately the try/except block wasn't catching the error on model instantiation so it appears down here as 'referenced before assignment'.

I'd double check the pycoral installation first and also the presence of model files in the models directory. I'm working on some better error checking code here so the error is more visible too. Will let you know once that's done.

I worked through this [Pycoral guide](https://coral.ai/docs/accelerator/get-started/#requirements) and managed to get it working on my laptop, after experiencing the same issues without pycoral installed correctly. 

Hope this helps! Let me know if there are further issues.

zerosnsa, 2023-12-12:
I am not using Google Coral USB, I am going to run the model directly on the Raspberry PI, is this OK? If it is shown in my python, this library is correctly installed

zerosnsa, 2023-12-12:
> 我现在添加了一些改进的错误检查逻辑。如果您想更新到最新版本，错误将更加清晰，因此希望修复也是如此。

[image]

geezacoleman, 2023-12-12:
You can run models directly on the Raspberry Pi but they will be quite slow without doing substantially more optimization. I wouldn't use a library like pycoral for that either. Unfortunately you can't just pip install pycoral - you need to follow the guide quite carefully for it to work. Even if it does appear in Python, it doesn't function correctly - it needs the Edge TPU runtime too.

If you update the OWL software to the latest version, the new error checking will provide you with the specific error.

Have a look at this repository for running models directly on the Pi 5: https://github.com/danigarci1/camera_tracking_rpi

zerosnsa, 2023-12-12:
So how should I run this project if I have jetson nano? What is the definition of slow time? What should my model staff look like if I use the jetson nano? The implementation seems to be more cost-effective than the jetson nano

geezacoleman, 2023-12-12:
If you use a Raspberry Pi 5 you could get higher framerates - it's hard to say exactly how fast it would be. But unoptimised object detectors on the Pi4 typically only get a 1-3 FPS. With the Google Coral you can get probably 15 - 20 FPS, or enough for reasonable operation speeds (of course depends on which model you choose - this is assuming a YOLOv8 N or similar). You can use pycoral without a Google Coral TPU connected, but you still need to install everything.

The project should work on the Jetson Nano fairly unchanged - but installation is more difficult. Setting up the Nano for model use requires some more work too, but again quite doable and it's something we've considered and would like to try. If you wanted to work on that and contribute it to the project that would be great!

zerosnsa, 2023-12-12:
So the jetson nano's frame rate is also indeterminate? May I ask whether the Connector-Panel Mount and Connector-Plug you use have the same function as the link below? They're supposed to function as a power switch, right? Is this link: https://detail.tmall.com/item.htm?id=744867676600

zerosnsa, 2023-12-12:
The main reason is that I can't buy these two connectors directly. If I buy them through international shopping and add the cost of Google Coral USB, the cost is much higher than that of jetson nano.

geezacoleman, 2023-12-12:
I'm being quite vague because it really depends on so many things:
* model size
* implementation
* training
* optimization

The Nano generally is more complicated to set up and unfortunately I couldn't offer any support. If you do get it working though it would be a really valuable contribution to the community! To help your decision, check out this [Reddit thread](https://www.reddit.com/r/computervision/comments/1161mab/should_i_get_a_google_coral_usb_accelerator_for/). It offers better advice than I could provide.

There are many guides out there for running YOLO and other object detectors on the Nano, so I think you'd be able to get help if you needed. You'll just need to get familiar with TensorRT and DeepStream.

> May I ask whether the Connector-Panel Mount and Connector-Plug you use have the same function as the link below? They're supposed to function as a power switch, right?

I unfortunately can't see what's at the link - do you have a screen shot?

zerosnsa, 2023-12-12:
[image]
Because I have not seen the specific function of the connector you used, I can't know whether the one in my country has the same function, please help me to have a look

zerosnsa, 2023-12-12:
I will try to run this project on jetson nano when I have time. Just because I have an idle jetson nano, I haven't run all the steps on Raspberry PI yet, so I can't directly try it on Raspberry PI. I want to start it when I understand the whole project

zerosnsa, 2023-12-12:
I will try to run this project on jetson nano when I have time. Just because I still have an idle jetson nano, I haven't run all the steps on Raspberry PI yet, so I can't directly try it on jetson nano. I want to start when I know the whole project

geezacoleman, 2023-12-12:
That connector looks quite good! You just need +/- 12V in and then a wire for every solenoid you want to run. So it would be 6 in the default OWL configuration. The one in the photo looks like it would work. If you want to use the 3D printed enclosure, just check the diameter and clearance of the panel mount part - ours is 30mm diameter.

Let me know how the Nano goes - will be great to see it run on that!

zerosnsa, 2023-12-12:
How many weeds does your model detect, and is the implementation affected by the size of the data set? Can models be shared? To simply run the model through Raspberry PI, do I just need to modify greenongreen.py, or do I also need to modify owl.py? At present, the layout of the model is a little confusing. I also hope that I can realize this on the jetson nano, which is more cost-effective for me

zerosnsa, 2023-12-12:
Hello, I still have a doubt, that is, the model reasoning is in Google Coral, then I use the Raspberry PI zero 2w and Raspberry PI 3 such as, the frame rate is not any difference?

geezacoleman, 2024-01-03:
> How many weeds does your model detect, and is the implementation affected by the size of the data set? Can models be shared?

The default detection is simply 'green' detection, so it will detect all sorts of weeds, plants and green objects. We don't currently provide any trained models for the 'green on green' or gog use. As mentioned in the response to #113, you just need to modify `owl.py`.

> the model reasoning is in Google Coral, then I use the Raspberry PI zero 2w and Raspberry PI 3 such as, the frame rate is not any difference?

I don't think that will work - the Pi3 and others are still slower and don't have USB3.0 support. [Check out this article](https://helloworld.co.in/article/google-coral-usb-accelerator-performance-raspberry-pi-3b-3a-4b) for more details.

[image]
[source](https://helloworld.co.in/article/google-coral-usb-accelerator-performance-raspberry-pi-3b-3a-4b)

zerosnsa, 2024-01-04:
[image]
Why does this error occur when I install Coral USB accelerator?

geezacoleman, 2024-01-04:
These installation instructions are taken from the official ones from Google/PyCoral. I double checked and they are the same, so unfortunately it is more of an issue for them or your own setup than the OWL.

But to start with I would double check you have internet access on the Pi and can access Google products/cloud services. The full set of instructions are here: https://coral.ai/docs/accelerator/get-started/#1-install-the-edge-tpu-runtime.

zerosnsa, 2024-01-04:
I use your system image, and according to the official steps to use, but I do not know where the problem occurred, there will be errors! You can ping Google on the Internet

zerosnsa, 2024-01-04:
I perform curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt - key add - when would be a mistake, according to can not find this, what is the version do you use

zerosnsa, 2024-01-04:
[image]
What I keep getting is this error, and I can't find libedgetpu1-std and libedgetpu1-max

geezacoleman, 2024-01-04:
The primary issue isn't that you can't find those two packages, it's that you cannot connect to packages.cloud.google.com where they are located. It's some type of permissions error where you're blocked from access.

Have a read through this forum - unfortunately I can't provide more help on this, so I would recommend researching the specific issue and using various forums and reporting back your solution.
https://www.googlecloudcommunity.com/gc/General-Misc-Q-A/How-do-I-unblock-my-server-s-IP/m-p/649978

zerosnsa, 2024-01-05:
[image]
I downloaded the libedgetpu1-std package offline and installed it. Then, if I used sudo apt-get install python3-pycoral, I could not find the pycoral package. I used pip3 to download it. However, in the owl environment, python has shown pycoral package, but when I execute the code, the following error occurs, indicating that there is no module, may I ask what is the problem?
[image]