geezacoleman/OpenWeedLocator #107: Power supply to RPi

MerbeinRobotics, 2023-12-15:
Hello.
I recently started field testing my OWL unit but have noticed it seems to "freeze" after about 90 seconds an then appears to 
 restart itself and go for another 90 seconds or so (I hear the beep beep start up sound with out switching the unit off and on) .  I suspect I have a power supply issue to the Raspberry Pi as the red power light seems to intermitantly turn off and on although the OWL still works even when the red power light is off.  I'm powering the RPi from a 12v battery and a 12v to 5v converter.

So I'm wondering if anyone else has experienced a similar problem and may have a fix?

Thanks,

Stephen

MerbeinRobotics, 2023-12-15:
Thanks Guy,

Yes I'll try powering the pi via the GPIO pins from the voltage regulator.  See photo of the setup below.  When I was building the system I had trouble powering the pi, the red light would not stay illuminated so I tried several regulators but changing the USB converter seemed to help the most.

I've included the photo of the pump set up just in case anyone might be interested in the plumbing as well.  Notice I had to put black tape over some of the green frame of the robot?  I had built the robot frame and painted it green before the OWL project came along.  When I first tested the OWL it was detecting the green paint of the robot, just as it should have!

I'll keep you updated.

PS Ever considered a PCB for the project?

Thanks,

KInd Regards,

Stephen  

[image]
[image]

MerbeinRobotics, 2023-12-16:
Just a quick update.  After bit more investigating I discovered the output pins on the voltage regulator I was using where loose and not soldered to the regulator's PCB properly.  After re-soldering it ran for a least 10 minutes so I think my problem is sorted.  Although the red power light on the RPi wasn't on.  Anyway I'll some more testing and see how it goes.

Thanks and Kind Regards,

Stephen

geezacoleman, 2023-12-16:
Thanks for the update Stephen - looks like you've found the issue. I've also had the painful problem of a broken wire inside the insulation before, which took about  a week to discover! They're some great pics of the rover, so feel free to share them on the Community Builds page too.

I'll close this issue for now but feel free to reopen if you need.

closed 2023-12-16 by geezacoleman