geezacoleman/OpenWeedLocator #109: [ERROR] Unknown algorithm gog, using default...

gnaegit, 2023-12-19:
I followed the instructions [here](https://github.com/geezacoleman/OpenWeedLocator/blob/main/models/README.md) to test the GoG algorithm. 
To install pycoral i followed the instructions [here](https://coral.ai/software/#debian-packages):
`python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0`
When running 
`(owl) pi@raspberrypi:~/owl $ python owl.py --show-display --algorithm gog`
I got the following output
'[INFO] Setting up nozzles...
[INFO] Nozzle setup complete. Initiating camera...
[INFO] Camera setup complete. Settings: 
Resolution: (416, 320)
Framerate: 40
Exposure Mode: beach
AutoWhiteBalance: auto
Exposure Compensation: -4
Sensor Mode: 0
 Nozzle 1    Nozzle 2    Nozzle 3    Nozzle 4 
                                              [ERROR] Unknown algorithm gog, using default...
[INFO] Using ssd_mobilenet_v2_coco_quant_postprocess_edgetpu model...

[ALGORITHM ERROR] Error while starting algorithm: gog.
Error message: Failed to load delegate from libedgetpu.so.1'
stating that the algorithm gog is unknown.

Is this issue known?

geezacoleman, 2023-12-19:
The error `[ERROR] Unknown algorithm gog, using default...` is a quirk of the way I've tried to better capture errors. It tried to look for a model file called `gog` but can't find it, so it then defaults to the first one in the directory. That isn't likely to be the error here, because it then loads `ssd_mobilenet_v2_coco_quant_postprocess_edgetpu` - is this the model you want to use?

`Failed to load delegate from libedgetpu.so.1` is likely causing the issues. I would recommend reading through this [tensorflow issue](https://github.com/tensorflow/tensorflow/issues/32743) and trying some of their solutions:
1. make sure the Google Coral is connected
2. restart the device after installing
3. changes to [udev rules](https://github.com/tensorflow/tensorflow/issues/32743#issuecomment-570811813) that may or may not have happened.

Let me know how this goes for you. I'll fix that `gog` error though.

gnaegit, 2023-12-19:
Thanks for the fast response and the guidance.
Unplugging and replugging the Google Coral did the job. I'm pretty sure the Google Coral was plugged in before, as the LED was turned on.

gnaegit, 2023-12-20:
We are from a Swiss university of applied sciences. We don't have any specific ideas yet and would like to use the system in research and teaching.

geezacoleman, 2023-12-20:
That's great to hear, let me know if there's anything I can do to help. There should be a lot more frequent updates and improvements next year.

closed 2023-12-29 by geezacoleman