geezacoleman/OpenWeedLocator #113: Boot item

zerosnsa, 2024-01-03:
Hello, I would like to ask if I need to modify the startup item when I use Green-on-Green, or do I not need to?

geezacoleman, 2024-01-03:
You won't need to change anything on the booting side, just the default setting of lines 521 - 522 of `owl.py`:

```python
ap.add_argument('--algorithm', type=str, default='gog', choices=['exg', 'nexg', 'exgr', 'maxg', 'exhsv', 'hsv', 'gog'])
ap.add_argument('--model-path', type=str, default='path_to_you_model', help='path to trained weed detection .tflite model or directory')
```

I realise it's quite a clunky/bad way of doing it currently, but the plan is to migrate to [`openweedlocator-tools`](https://pypi.org/project/openweedlocator-tools/0.0.1/) package this year, which will move all these settings into a `config.json` file you can more easily edit.

zerosnsa, 2024-01-24:
Hello, why do I want to do an operation to modify the sequence of the nozzle? Why does he show that there is no print after I make a judgment
[image]

geezacoleman, 2024-02-08:
Closing this as completed - feel free to reopen of needed.

closed 2024-02-08 by geezacoleman