geezacoleman/OpenWeedLocator #116: Field Start-Up

Kmrfarmsinc, 2024-01-11:
Hello again, I am at the very last step until I take it out in the field for testing. The only problem that I think will be an issue is the program running as soon as I put power to it. When I want to run the owl program on my desktop, I have to execute that in the terminal. How would you adjust it so it starts working right away when I get in the field? 

I looked through the previous issues and didn’t seem like it was something talked about.

geezacoleman, 2024-01-12:
Great that it's almost ready for field testing.

The OWL software should start automatically on powerup. Do you hear two beeps when you power it on without anything attached? 

All the hassle with stopping programs just to start them again is because of owl.py starting when the Raspberry Pi boots. If it's set up correctly you should be able to turn it on/off and hear the beeps without using the terminal.

If that doesn't happen, it might be because a couple of files aren't set to executable. You can run these lines from terminal and it should fix it:

```
(owl) pi@raspberrypi:~/owl $ chmod a+x owl.py
(owl) pi@raspberrypi:~/owl $ chmod a+x owl_boot.sh
```

The only changes you'll need to make to owl.py file are if you have changed default settings through the command line. For example if you used --exp-compenstion 0, that won't be saved. So you need to go into the code and change the default.

geezacoleman, 2024-02-08:
How did you go with this @Kmrfarmsinc? Let me know if you run into any other issues and I can try to help.

Kmrfarmsinc, 2024-03-14:
I have been off and on trying to get it going but I still am having trouble with the auto start. I have a buzzer and when I first turn on the raspberry pi, I only get one beep. I’m not sure what this means. I don’t know if the program is just supposed to be running in the background for the auto start or if you are supposed to see the feed like normal. Everything else seems to be working well. I have a quality video, and I see the nozzles that are wanting to spray. If you have any suggestions on what the problems may be, let me know when you are able to. Thanks

Kmrfarmsinc, 2024-03-15:
I tried the initial two lines that you suggested and that didn’t seem to work. I looked in the detailed startup and tried to use those 5 lines to have that work, and that didn’t work as well. Just wanted to give more info on what I did. (Feed means video feed, show we always see that when it’s working or could it just be in the background?)

Kmrfarmsinc, 2024-03-16:
So I just got it to work with the two beeps at startup and the LED light showing me when it is spraying. The picture is clear and everything seems to be working perfectly. The program does seem to be working in the background off start-up because of the LED flashing. I did buy your relay board because you made a really good board (probably show advertise that a lot more in the instruction page instead of just the forums), and it was easy to get everything set up. When I use it this season, I will be sure to send videos of the program running back here for people to see the benefits from it, thanks again.

geezacoleman, 2024-03-19:
That's great, glad you could get it working. If everything is set up correctly, it will just work in the background after start up with no video feed or any visible signs, besides the LED flashes/solenoid clicks you might hear. Two beeps after startup is essential - if you only hear one or none, then it probably isn't working correctly. 

Just for the benefit of others reading this, what do you think the issue was and how did you fix it?

Thanks for purchasing those OWL driver boards too, they should make it a bit easier. How have you found them so far?

Kmrfarmsinc, 2024-03-28:
[image]

As far as the problems with setting up the program, that was all user error being the first time I’ve tried something like this. The hardware was a bit tricky but if we bought all of your products you suggested it would have worked fine. We just miss wired the onto the prongs the first time. 

We just got the new board, and haven’t tried it yet. Does the wire schematic that I drew look correct. It says 5v on the PC board. I’m assuming that comes from the PI computer and we use 12v to power up the solenoids? The jumper I have circled stays non-jumped? I just could find any instructions on this and didn’t want to wire it wrong.