geezacoleman/OpenWeedLocator #118: Model deployment

zerosnsa, 2024-01-24:
I am training the model according to the example, but I keep showing this error when I deploy it. The model works fine if I use my own computer locally
[image]

zerosnsa, 2024-01-24:
I used the generic model and it worked fine on the Raspberry PI

geezacoleman, 2024-01-24:
Awesome that you got the YOLOv8-N working correctly at 416 x 320 resolution. Do you mind sharing how you did that? I've read that it would only work at 192 x 192 resolution.

I found this [issue ](https://github.com/google-coral/edgetpu/issues/11)on the pycoral page about the error you receive. It appears it's a power supply issue? Do you have a powered USB hub available?

zerosnsa, 2024-01-24:
But did I just run to report this error? And when I run the sample model, it works fine, but when I run my own trained model, this error occurs.

zerosnsa, 2024-01-24:
Isn't this resolution set in the code? I didn't change the add resolution operation

geezacoleman, 2024-01-24:
Probably a few issues here - which sample model are you using? and what resolution are you training your custom model at? You should make sure the resolution set with owl.py matches the trained model resolution.

There are issues exporting from YOLOv8 using their export function, unless you train and set it to 192 x 192.