geezacoleman/OpenWeedLocator #128: Compact OWL model model defects / printability issues

MisterHW, 2024-05-13:
Compact OWL as of https://github.com/geezacoleman/OpenWeedLocator/tree/68f2ee177ff4bf59b490357e216150cbcec619c8/3D%20Models/Compact%20OWL has a few issues affecting printability and the function of the o-ring face seals.

### tray.stl

The bottom of the model is not flat, leading to most of the area not being exactly on the print bed

### main-body.stl

The profile presents as a multi-part stl. Older slicers don't fuse everything into a single object, which causes the rails to be pinted separately, and they may or may not be bonded to the profile.

[image]

failed test print section:

[image]

### frontplate.stl

The o-ring groove does not follow the main body cross-section.

[image]

## partial recreation in FreeCAD

FreeCAD may not be the best choice for a redo, but I've recreated the extrusion profile and a blind end cap (which can then be cloned and specialized). This ought to give a glimpse into the workflow and level of hassle when using this particular CAD program. 

[OWL Compact FreeCAD example (plus 3mf, step).zip](https://github.com/geezacoleman/OpenWeedLocator/files/15298824/OWL.Compact.FreeCAD.example.plus.3mf.step.zip)

[image]

Here's another view of the end cap with o-ring groove. It's not dimensioned absolutely correctly. 

Note: 3D printed parts can be expected to exhibit some level of creep, especially since there is a heat source inside. Therefore, it's probably best to use Shore 15 EPDM foam o-ring material and a conservative amount of compression. Since the flat areas around the M5 holes butt up against ```main body```, it's the depth of the groove w.r.t. this plane.

[image]

geezacoleman, 2024-07-03:
Thanks for finding these @MisterHW ! Not sure how those rails separated in the design, but they should be fixed now in #137. I have also added a blank backplate and raised up the receptacle/gland to avoid contact with the USB/ethernet ports on the Raspberry Pi when fitted.

MisterHW, 2024-07-04:
Thanks for addressing it. In a future PR, a few other tasks can be addressed:
- completed re-creation of the Compact OWL design in FreeCAD
- .step export of components and a full model
- a visual index markdown showing all the models with explicit instructions which files to download and print

closed 2024-07-04 by MisterHW