geezacoleman/OpenWeedLocator #142: Green on green

bassyboi, 2024-08-24:
Trying to figure out how to install coral on raspberrypi 4 using python 3.12!!! I tried back dating the python but run into trying to troubleshoot why the code not running with everything required and paths set up correctly!!! Any suggestions?? I have a data set here that seems to work well in pasture on laptop on my own script over a video!!  Any suggestions

geezacoleman, 2024-08-26:
Thanks for giving this a go! As you've probably found out unfortunately Google has not updated the PyCoral library for the Coral in some time. There are a few options you can try that have been developed by the community. What are your specific errors? Can you screenshot/copy paste them here? Which dataset are you using or is it one you've developed yourself?

There's a great video explaining the set up process by the [YouTuber Jung](https://www.youtube.com/watch?v=HMeDLSflCA0). This thread on the issue (https://github.com/google-coral/pycoral/issues/137) also provides some options to run it with post 3.9 versions of Python. The issue with backdating python is you'll run into compatibility issues with picamera2 I think and other packages.

I explain a bit more about the situation/plans in #141. When I set up GoG support for the OWL with the Coral and PyCoral I assumed there would be ongoing support for the hardware/software. Unfortunately it seems that isn't the case and the community is moving toward other options like the Hailo AI Kit. The Raspberry Pi 5 can run small models at moderate FPS, but having the additional hardware like the Hailo chip using the PCIe connection greatly improves speed. I'd recommend upgrading to a Pi 5 if you want to run GoG models in the future - it will be much better supported/faster.

bassyboi, 2024-08-26:
I have tried back dating the raspberry pi!!! I kinda wrote a script that sets up setting up python 3.9 in its own environment to run the coral (env-coral) in and use the rest of owl to run in the owl in (owl) I got it running once thought I got it right had camera up detecting weeds in my lawn thought I solved it then reset raspberry pi and kept saying I have to install open cv and a few and can’t get it to run again

bassyboi, 2024-08-26:
Also yes I made a custom data set

closed 2024-08-28 by bassyboi

geezacoleman, 2024-08-30:
This is really neat! Thank you for sharing the process and my apologies for not responding sooner or for being much help! It looks like you've got it sorted far better than I could have in any case. Your dataset sounds great - I have found that grouping very different weeds together may negatively impact model training, because of the diversity in appearance that the weeds present. Depends on the context and crop/background however. I have also found the opposite, where too many classes of very similar classes can result in poorer training, so I think there is a middle ground. Do you have some examples?

GoG will be a focus for me in the next 6 months starting from late September, so expect there to be a lot more support for it and updates. The idea is to update the [openweedlocator-tools](https://github.com/geezacoleman/openweedlocator-tools) package and use that as the base for the main OWL repository. That way functions can be run independently and interact better with a GUI too.

Thanks again for sharing!

bassyboi, 2024-08-30:
yeah i trained and retrained the model with the likes clover mash mellow weed  and that for a trial... just with weeds and that around the house in my area but had like 1500 photos and i hated sitting there trying to put a box over every weed haha