geezacoleman/OpenWeedLocator #143: Consulta

Kapo030, 2024-08-26:
Hola.. como puedo lograr armar un equipo de estos? Soy de argentina

geezacoleman, 2024-08-27:
Hey! Thanks for the question - how many OWLs did you want to use together? Each OWL operates individually, so it is possible to use lots of OWLs together on one sprayer, but controlling them is more difficult. The independence of each device makes it difficult to monitor for errors or to change detection mode, so we're in the process of developing an in-cab controller so wider sprayers can be fitted.

Hopefully I understood your question correctly!

geezacoleman, 2024-11-14:
I'll close this out now, but feel free to post any further questions/comments as needed.

closed 2024-11-14 by geezacoleman