geezacoleman/OpenWeedLocator #145: flush system

bassyboi, 2024-08-31:
just wondering if maybe adding a flush system when dealing with chemicals to prime up chemicals and flush out chemicals for changing chemicals and cleaning boom for components longevity...

geezacoleman, 2024-09-01:
The new controller now includes an option to [turn all solenoids on](https://github.com/geezacoleman/OpenWeedLocator/blob/525912dffba40391ce3ac0e504f064b3766da9fb/utils/input_manager.py#L132) which could be used for flushing. Recirculating booms are quite common for the reasons you mention too.

Is that what you're thinking of?

bassyboi, 2024-09-01:
yeah neat i was thinking maybe keyboard presses but like b button to start flush on timer and e to stop if need

geezacoleman, 2024-09-18:
Ah I see, yeh that makes sense. Pretty easy to add if you think it's worthwhile? I can put in a 'flush' option as you mention if you press 'f' on the keyboard. Useful if you have a keyboard out in the field I guess! Will be on the all the controllers as well.