geezacoleman/OpenWeedLocator #146: green on green lidar or height adjustment

bassyboi, 2024-08-31:
keeping boom at correct height in gog applications to allow correct distant for boom height and camera field of view...

geezacoleman, 2024-09-01:
Yes, this is definitely a challenge. My feeling is lidar might be overkill for it? Perhaps a ToF sensor or ultrasonic sensor are the easiest ways to do it. I think commercial booms use ultrasonic sensors spaced out over the length for this.

bassyboi, 2024-09-01:
ill checkout how the new jd ploughs do it have used one that is called accudepth i will investigate

bassyboi, 2024-09-01:
TF-luna Lidar Sensor 0.2m-8m maybe something like this +- 6cm accuracy 

i know holley now make a similar sensor for there new fuel tank gauge that is lidar  the new jd ploughs do have them and they work great

geezacoleman, 2024-09-18:
This is pretty neat and quite cost effective too. How would you control boom height with this? Alternatively if you know boom height at different locations, you can better map detections to actual location, rather than the basic way it works currently.

bassyboi, 2024-09-18:
I seen something even neater then this with object detection and depth cameras might be expensive!!! 

I’d just set a actuator up or solenoids up and down 50 mm+-  threshold but my still need to think more about this