geezacoleman/OpenWeedLocator #174: Create “canbus” Branch for Basic Section ON/OFF via CAN with Node‑RED (Master)

bassyboi, 2025-04-07:
Description:

We propose creating a dedicated “canbus” branch in our repository to consolidate all work related to CAN bus integration. This branch will include our initial implementation of a basic Node‑RED master running on a Raspberry Pi that uses GPIO inputs to send CAN commands to control sections (meters) by turning them on or off. Error messaging and advanced handling will be addressed in future iterations.

Goals for the “canbus” Branch:
	•	Basic Functionality:
	•	Implement a Node‑RED flow for the master Raspberry Pi that uses GPIO input nodes (rpi gpio in) to detect button presses.
	•	Convert GPIO events to JSON payloads indicating the specific section (meter) and desired command (1 for ON, 0 for OFF).
	•	Build a CAN message (for example, using CAN ID 0x300 with a two‑byte payload) and transmit it via a SocketCAN Out node.
	•	Demonstrate that sections can be individually turned on or off, or that broadcast commands (e.g., ALL ON/OFF) work as expected.
	•	System Integration:
	•	The basic CAN messages from the master will be used to control slave nodes (each representing a one‑meter section) which will eventually integrate with Open AgiGPS.
	•	The focus of this branch is on the master’s ability to send correct commands via CAN; error messaging, logging, and advanced handling will be implemented later.

Motivation:
	•	Modular Development:
Creating the “canbus” branch allows us to keep our CAN bus integration separate from our main codebase while we work on and refine this feature.
	•	Basic Control Implementation:
Initially, we need only a straightforward mechanism that uses physical buttons (via GPIO) to trigger CAN messages to turn sections on or off.
	•	Future Enhancements:
Once the core functionality is stable, we can integrate enhanced error messaging, robust logging, and advanced command processing.

Acceptance Criteria:
	•	A new branch named “canbus” exists containing:
	•	A Node‑RED flow configured on the master Raspberry Pi that reads GPIO inputs.
	•	Logic to convert inputs into JSON payloads for the CAN bus.
	•	A function to construct and send a CAN message (using CAN ID 0x300) via SocketCAN.
	•	Functionality that correctly turns sections on or off when corresponding hardware buttons are actuated.
	•	Documentation is updated with instructions for setting up Node‑RED on the Raspberry Pi and testing the basic GPIO-to-CAN integration.
	•	Future tasks for error handling and advanced features are identified and documented for later development.

Additional Context:
	•	This is a basic implementation aimed at establishing reliable ON/OFF control. Error handling and more detailed logging messages will be developed in a subsequent iteration.
	•	We currently use Node‑RED on the master Raspberry Pi as our control hub. Slave nodes (one‑meter sections) will receive the CAN commands and interface with Open AgiGPS in later branches.
	•	Any related issues or pull requests should reference this branch for continuity.

bassyboi, 2025-04-08:
https://github.com/bassyboi/Owl-can-master-/tree/main

Still got to write read me.md sort of a draft read me also need to make the install better with the right libraries and that for it to work! Will update

EchelonMirror, 2025-04-28:
You thinking of running this virtually or having literal CAN hardware on the units? Alternative consideration for powering units being PoE/PoE+ and having all comms over a higher throughout bus.